cyborg5/IRLib2 #3: Support for RC6 36 bits

cklim01, 2016-09-29:
Do this new library support to send and receive RC6 36 bits code?

cyborg5, 2016-09-29:
We support a variety of RC6 variations including a 32 bit MCE variety. Depending on you count the bits I can see how you could describe it as 35 bits but I am unaware of any 36 bit versions. If you can provide me a description on how it works perhaps using IRP notation or any other kind of specification we could try to support it. Can you point me to a webpage perhaps describing this variety?

cklim01, 2016-09-29:
Chris, actually i am refer to http://www.righto.com/2010/12/64-bit-rc6-codes-arduino-and-xbox.html. I have a satellite tvbox which use RC6 36bit. I am newbie in mcu and programming. I wanted to do a Bluetooth remote control using Attiny85.

cyborg5, 2016-09-29:
I'm about 99% sure that the RC6-6-32 also known as MCE is what Ken is calling 36 bit RC6. While researching this actually discovered a problem with my documentation. I've incorrectly identified which bit of this protocol is the title bit. I said that it was the 32nd most significant bit and actually it's the 16th most significant bit. It doesn't affect how my code works but it does affect my documentation and some of the sample code that uses toggle bits.

You mentioned Bluetooth and ATtiny85. Note that my library is for infrared remotes only and has nothing to do with Bluetooth. Also I doubt that you could support Bluetooth on that kind of processor because it is so limited. If you're interested in Bluetooth I suggest you check out Adafruit.com especially the feather boards that can do Bluetooth. I don't know that they have in the sample code for controlling Xbox via Bluetooth but I would check with them first.

I'm going to leave this issue open until I'm able to update the documentation.

Do you have an infrared remote that will control an Xbox?