cyborg5/IRLib2 #4: How to paste :/

VytautsDV, 2016-10-02:
Hi, nice library, receiving works great, but i just dont get it how to send raw data, im a little bit confused, need help.

i have this 
=================================================================
Ready to receive IR signals
Do a cut-and-paste of the following lines into the 
designated location in rawSend.ino

#define RAW_DATA_LEN 68
uint16_t rawData[RAW_DATA_LEN]={
	4400, 4450, 550, 1650, 550, 1650, 550, 1650, 
	500, 550, 550, 550, 500, 550, 550, 500, 
	550, 550, 500, 1700, 500, 1700, 500, 1650, 
	550, 550, 550, 500, 550, 500, 550, 550, 
	550, 500, 550, 550, 500, 1700, 500, 550, 
	500, 550, 550, 550, 500, 550, 550, 500, 
	550, 550, 500, 1700, 500, 550, 500, 1700, 
	500, 1700, 550, 1650, 500, 1700, 500, 1700, 
	500, 1700, 500, 1000};

============================================

where and how much do i paste of this in here 

(whole method)

-----------------------------------------------------------------------------------------
void irTransmitter() {
  if (bttnSTATE == HIGH) {
    myRawSender.send(rawData,RAW_DATA_LEN,36);
    Serial.println(F("Sent signal."));
  }
}
------------------------------------------------------------------------------------------

could you do this?

cyborg5, 2016-10-02:
Everything from the #define that you have listed there down through the… 1000); should be cut and pasted into the rawSend.ino sketch at approximately the line 22. See the comments in the rawSend example. Also see section 2.3.3 in the users manual.

You can modify that sketch to work off of a pushbutton instead of the serial monitor as you have shown. As to where to pasted in that particular sketch that you have shown, it depends on where you want the data to be assessable. You can post it globally in the sketch or you could put it inside your right before your if statement.

VytautasDV, 2016-10-02:
myRawSender.send(rawData, RAW_DATA_LEN, 68);

could you explain me this line, what is rawData. im geting:

exit status 1
'rawData' was not declared in this scope

VytautasDV, 2016-10-02:
oh. stupid me. i was trying to paste the code in method field @ myRawSender.send(**rawData, RAW_DATA_LEN, 68**);

VytautasDV, 2016-10-02:
@cyborg5 what pin to send should i use on MEGA. yesterday when i was tired i remember reading its 3 pin, but now when i search i cant find it.

cyborg5, 2016-10-02:
See the table in the users manual section 1.5.1 reveals the default pin number for a mega is pin 9.

VytautasDV, 2016-10-02:
yeah sorry for stupid questions. i opened and read through the manual again.

VytautasDV, 2016-10-02:
@cyborg5 

well it seams to me my code is good, but now a problem. It doesnt seem to work at all. im not using any ir LED driver, maybe thats my mistake?

i did set up my system so that receiver reads my transmitter. And it seems that transmitter only sends like 3-4 values instead of 68.

closed 2024-03-07 by Zuzuske