cyborg5/IRLib2 #5: Tiny repeater

sid1202, 2016-10-19:
I'd like to build a dedicated IR repeater using a Adafruit 5V trinket (AtTiny85) but I did not see any info the this chip in the manual. 

The satellite controller is in the living room. The output is split with a TV in the bedroom (one wall separates the 2)

I have a prototype running on a Uno. Any pointers to get this running on a trinket ?

Note: The size of the trinket is what makes it appealing for this project, I can hide it easily.

Thank you,
Sidney

cyborg5, 2016-10-19:
We currently do not support the ATtiny85 but we are looking into it. Even if we can't get it to work with IRLib2 it might only work for sending only or receiving only but not both and it would only be able to use at most one or two protocols because that chip has so little memory available. My recommendation would be the Adafruit Pro trinket https://www.adafruit.com/products/2000 which is based on the same processor as the Uno. Or better yet the Adafruit Feather 32u4 basic prototype https://www.adafruit.com/products/2771 it's a little more expensive but it has a built-in Lipo charger and a nice prototyping area where you could put your LED driver circuitry and your receiver device.

sid1202, 2016-10-19:
Since this library is modular I hoped that by having only one protocol both receive and send would work. might need to eliminate the bootloader though. There is a small decoration on the wall in question and the trinket barely fits in and there is a place to put the receiver so it won't show. Since I could not get the IR repeater going on the Trinket I tried to fit a Pro-Trinket, but it's to wide and too long. Its something for my mother so I can't hack the decoration in anyway either.

I'll set a Pro-Tricket up until I can get it running on the trinket if she'll let me lol

Shears and great work. Congratulations!

Sid