cyborg5/IRLib2 #6: Smallest possible footprint

supuflounder, 2016-10-20:
I need the smallest footprint possible so I can fit this into a small machine.  I do not need to read or send any known protocol.  What I need to be able to do is send 16-bit data packets, with a checksum (CRC-16) in the last one.  Somewhere between five and ten data packets, then an ACK/NAK to indicate if the checksum matched the data.  The first 8 bits will indicate the total number of packets that will be sent.  There will be two kinds of ACK packets, both of which are a single 16-bit packet.  If the first byte of the packet is NAK, retransmission will have to take place.  If the first byte is ACK, the message has been receied sucessfully.  The sender is ultra-low-power.  If the second byte is 0, the transmitter computer shuts down until its next scheduled event occurs.  If the second byte is 1, the transmitter will listen for data to be sent, and process that data, and shut down until it is again needed.  It normally wakes up every minute or two, reads some sensors, and if they suggest action is required, then a message is sent via IRLIB to a computer without a power budget.  What do you suggest as the smallest-footprint approach?

cyborg5, 2016-10-20:
I don't know that I can really recommend any of hardware design. I would look at the Adafruit feather 32u4 or the Adafruit Pro Trinket for an off-the-shelf solution but anything smaller you would have to design your own circuitry. We do not support the ATtiny85 so a regular Adafruit trinket is out of the question.

supuflounder, 2016-10-25:
> I don't know that I can really recommend any of hardware design. I would
> look at the Adafruit feather 32u4 or the Adafruit Pro Trinket for an
> off-the-shelf solution but anything smaller you would [...]

The chip we are using is the Tiny167.  I am using 14K of its 16K, and I
have identified 5K of USB-related fat I hope I can trim.  We have no USB
port because we have no board space.  We cannot use commercial boards
because of the size; our entire device, including battery, fits in a 1"
cylinder, 5/8" high.  I was not asking anything about hardware design; I
was asking how to configure the library for the smallest possible
footprint. I only have 16K of flash.
       Joe