cyborg5/IRLib2 #7: common anode LED pair

supuflounder, 2016-10-20:
There are a couple instances of dual-emitter LEDs.  For example, 660nm (red) and 940nm (IR).  The problem is that these come in 3-lead common-anode packages.  So to turn the IR diode "on" requires setting the output pin "low", not "high".  How difficult do you think it would be to make this change?  I don't want to get 3/4 of the mods in only to discover that the remaining 20% of the changes are going to use the other 80% of the time.  We have zero board space left, so the space savings from moving from two discrete 3mm emitters to one combined package actuallyf
    thanks.    
    joe

cyborg5, 2016-10-20:
I'm a bit embarrassed to admit that the hardware specific parts of the code are almost beyond my understanding. They are a slightly rearranged version of the code from Ken Shirriff's IRremote library. The code involves manipulating hardware registers to set up a specialized frequency adjustable PWM using hardware timers. You can look at IRLibProtocols/IRLibHardware.h at about line number 174 assuming you're using the default TIMER2 and see if you can figure out what you need to do. The value IR_SEND_PWM_START turns on the modulated signal and IR_SEND_PWM_STOP turns it all off but I'm not sure how you get the start version to conclude with the pin high unless you just stuck in your statement like digitalWrite(pin, HIGH). You might do better to use the bit-bang version of sending which starts at line 148. Just reverse the high and low statements in the digitalWrite commands. I'm assuming you're not using any kind of driver circuit because you said your board space was extremely tight. If you are using a transistor to drive the LED you could reverse the logic by substituting PNP instead of NPN.

supuflounder, 2016-10-25:
Thanks for the reply.  I looked at that bit-bang code, and I have not
programmed that close to the metal in decades, and didn't want to attempt
it on an unfamiliar architecture (I, who used to write chained
channel-command words (CCWs) on IBM mainframes, and manipulate the bits in
the I/O controllers from the original 8088 PC through the 80386).  I can
probably figure it out, given enough time, but time is in short supply as
we have an Oct 31 deadline.  Because the IR path will be under three feet,
and possibly as little as 3 inches, power was not a primary concern,
hence, the decision to drive the IR emitter directly from the processor.

The tiny board size we have (two single-sided boards, 1" diameter  and
including battery, 5/8" high), being able to reduce from four components
to one component matters, thus the desirability of the dual-emitter unit. 
If only they made a common-cathode version!
      Joe