cyborg5/IRLib2 #13: Lego PowerFunctions protocol

UgaDaBuga, 2016-12-01:
Hello.
I am new here and i dont even think this is the right way to post things.
I spent last 3-4 days digging teh net for a library for arduino that understands my lego IR transmiters.
Turns out nothing out there that is available works. Codes i get from my IR are plain bullshit, bits dont fit anything.Apparently the decode function from older libs doesnt understand my Lego IR pilots.

As I understand, this library is an "evolved" version of what i tried to use. 
Can anybody "include" Lego RC protocol in this library so i can finaly read some commands from lego pilot in my arduino ? I dont think i understand all the "encoding" and "us" ms to implement.....

Sample protocol description from LEGO:
http://storage.technicbricks.com/Media/2010/TBs_20100304_1/LEGO%20Power%20Functions%20RC%20v120.pdf
I couldnt find any newer i wonder if lego stopped publishing that document.

Would be cool if somebody answered...

penfold42, 2016-12-01:
It looks like a variable space encoding similar to the NEC protocol but with different timing.

UgaDaBuga, 2016-12-08:
I dont think this is the case. I dig around the whole lib and keep getting crap no matter what i do. This is way over my head. I created a decoder etc. Whatever my controllers send is unreadable by this lib. All I get some  6 numbers just like the lego protocol says then some crap follows,Just they dont match anything.

I found this:
There is another protocol that we are aware of that we have not yet supported that uses a completely different system. The Async protocol sends data as if the IR LED is connected directly to a UART sending asynchronous serial data. A value of one as indicated by a mark in a value of zero is indicated by a space. Typically it begins with a start bit followed by 8 or 16 databits perhaps followed by a parity bit. So for example an 8-bit value of zero would be a mark followed by 8 time periods of space followed by a parity bit depending on whether you use odd or even parity. Similarly an eight bit value of 255 would be nine time periods of mark followed by a parity bit that might be a mark or a space. We may attempt to create a decoder for this protocol in a future version.

Is this the case with lego pilots?
How do i proceed with that? 
How to detect something like this:
To ensure correct detection of IR messages six 38k Hz cycles are transmitted as mark. Low bit consists of 6 cycles of IR and 10 “cycles” of pause, high bit of 6 cycles IR and 21 “cycles” of pause and start bit of 6 cycles IR and 39 “cycles” of pause.