cyborg5/IRLib2 #14: Integer overflow in IRLib_P08_Samsung36.h

JohnWasser, 2016-12-09:
```In file included from /Users/john/Documents/Arduino/libraries/IRLib2/IRLibAll.h:31:0,
                 from /Users/john/Documents/Arduino/libraries/IRLib2/examples/dump/dump.ino:9:
/Users/john/Documents/Arduino/libraries/IRLibProtocols/IRLib_P08_Samsung36.h: In member function 'void IRsendSamsung36::send(uint32_t, uint32_t)':
/Users/john/Documents/Arduino/libraries/IRLibProtocols/IRLib_P08_Samsung36.h:53:12: warning: integer overflow in expression [-Woverflow]
   space(118*500);    //Lead out is 118 times the base time 500
            ^
```
118 * 500 = 59000  Too big for the default signed integer.  Suggest changing to:
`   space(118U*500U);    //Lead out is 118 times the base time 500`

pandaiolo, 2017-07-17:
The fix is working for me, thanks

cyborg5, 2017-07-17:
Sorry I failed to notice this issue several months ago. I was in the hospital at the time. Fixed in the latest update.

closed 2017-07-17 by cyborg5