cyborg5/IRLib2 #17: Incompatibility between FastLED lib and IRLib2

tweep, 2017-02-06:
Heya, 

thanks for such a great library - I'm  using it to decode IR signals to remotely control LED, which are driven by the FastLED library ( https://github.com/FastLED/FastLED). 

All decoding works fine for me **until call the FastLED.delay()** function to set the frame-rate of the LEDS - after calling FastLED.delay(40), the signals of the IR receiver are not decoded correctly anymore ( they change everytime i press the same button on the control).

tweep, 2017-02-06:
Hi, 
looking at the FastLED issues this one came up which describes problems between some LED chipsets ( like the WS2811 I am using ) and timings. 

> **The timing protocol for WS2812 leds requires disabling interrupts while writing out led data, and the IR library relies on interrupts to read the IR data, which means that if you're using the remote while the library is writing out led data, then it is going to miss part of the ir signal and misinterpret it.**

https://github.com/FastLED/FastLED/issues/198

I will go and try VorpalLemur's suggestion: 

`IRrecv irrecv(RECV_PIN);
decode_results results;

loop() {

    /* do effects here */

    while (!irrecv.isIdle());  // if not idle, wait till complete

    if (irrecv.decode(&results)) {  
        /* respond to button */

        irrecv.resume(); // Set up IR to receive next value.
    }

    FastLED.show(); 
}`

curlyz, 2017-07-06:
WS2812 and IR signal are so time sensitive that you should not put WS in the loop, what you should do is disable everything, receive IR, and then set WS