cyborg5/IRLib2 #22: Invalid signal disables RC decoding ability permanently

hazarkarabay, 2017-04-08:
Hi, 

I'm using IRLib2 for RC6 decoding. When it works, it works beautifully, so first of all thanks for the hard work!

I have a CCFL desk lamp which have ability to break IR decoding. When it's first starting up, it can generate some garbage on my TSOP38238's output (checked with a scope). Sometimes, this garbage signal can completely break IRLib2's decoding ability. 

Test sketch:
```
/* Includes */
#include <avr/wdt.h>
#include <IRLibRecv.h>
#include <IRLibDecodeBase.h>
#include <IRLib_P04_RC6.h>
#include <IRLibCombo.h>

/* Library init */
IRrecv myReceiver(2);  // pin number for the IR receiver
IRdecode myDecoder;

void setup() {
  // Setup WDT
  wdt_enable(WDTO_2S);
  myReceiver.enableIRIn(); // Start the receiver
  Serial.begin(115200);
}

void loop() {
  // IR receive and decode stuff
  if (myReceiver.getResults()) {
    if (myDecoder.decode()) {
      Serial.println(myDecoder.value);
    }

    myReceiver.enableIRIn();
    myDecoder.dumpResults(true);
  }
}
```

Output when this issue happens:
```
Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:1838
  Head: m14546  s1838
0:m6354 s-22738	1:m-26414 s-14546		 2:m22738 s1838	3:m-26414 s-6354		 
4:m30930 s1838	5:m-1838 s-6354		 6:m-18222 s18222	7:m-26414 s-14546		 
8:m-18222 s-30930	9:m-18222 s-6354		 10:m30930 s-6354	11:m30930 s1838		 
12:m6354 s-22738	13:m6354 s-6354		 14:m22738 s18222	15:m-1838 s-30930		 

16:m-1838 s-30930	17:m-10030 s-14546		 18:m-10030 s26414	19:m-18222 s-30930		 
20:m-1838 s-22738	21:m-18222 s-30930		 22:m30930 s26414	23:m-18222 s1838		 
24:m-10030 s26414	25:m-26414 s-22738		 26:m14546 s-6354	27:m30930 s-22738		 
28:m14546 s-14546	29:m-26414 s1838		 30:m30930 s1838	31:m30930 s10030		 

32:m6354 s18222	33:m14546 s18222		 34:m-18222 s-30930	35:m30930 s1838		 
36:m-26414 s-6354	37:m14546 s26414		 38:m-10030 s-14546	39:m30930 s-30930		 
40:m30930 s-22738	41:m-26414 s18222		 42:m-10030 s-6354	43:m-18222 s26414		 
44:m22738 s1838	45:m30930 s18222		 46:m-26414 s1838	47:m-10030 s26414		 

48:m-26414
Extent=4294842578
Mark  min:-26414	 max:30930
Space min:1838	 max:26414

Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:26414
  Head: m6354  s26414
0:m-10030 s-22738	1:m-10030 s-6354		 2:m22738 s26414	3:m-10030 s10030		 
4:m-26414 s26414	5:m-26414 s10030		 6:m6354 s-6354	7:m-10030 s-6354		 
8:m6354 s26414	9:m6354 s10030		 10:m-26414 s10030	11:m-26414 s26414		 
12:m-10030 s-22738	13:m-10030 s10030		 14:m22738 s-6354	15:m-26414 s26414		 

16:m-26414 s26414	17:m22738 s-6354		 18:m22738 s10030	19:m6354 s26414		 
20:m-26414 s-22738	21:m6354 s26414		 22:m-26414 s10030	23:m6354 s26414		 
24:m22738 s10030	25:m-10030 s-22738		 26:m6354 s10030	27:m-26414 s-22738		 
28:m6354 s-6354	29:m-10030 s26414		 30:m-26414 s26414	31:m-26414 s-22738		 

32:m-10030 s-6354	33:m6354 s-6354		 34:m6354 s26414	35:m-26414 s26414		 
36:m-10030 s10030	37:m6354 s10030		 38:m22738 s-6354	39:m-26414 s26414		 
40:m-26414 s-22738	41:m-10030 s-6354		 42:m22738 s10030	43:m6354 s10030		 
44:m22738 s26414	45:m-26414 s-6354		 46:m-10030 s26414	47:m22738 s10030		 

48:m-10030
Extent=121042
Mark  min:-26414	 max:22738
Space min:10030	 max:26414

Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:10030
  Head: m55506  s10030
0:m22738 s-22738	1:m22738 s10030		 2:m22738 s10030	3:m22738 s-22738		 
4:m-10030 s10030	5:m-10030 s-22738		 6:m-10030 s10030	7:m22738 s10030		 
8:m-10030 s10030	9:m-10030 s-22738		 10:m-10030 s-22738	11:m-10030 s10030		 
12:m22738 s-22738	13:m22738 s-22738		 14:m22738 s10030	15:m-10030 s10030		 

16:m-10030 s10030	17:m22738 s10030		 18:m22738 s-22738	19:m-10030 s10030		 
20:m-10030 s-22738	21:m-10030 s10030		 22:m-10030 s-22738	23:m-10030 s10030		 
24:m22738 s-22738	25:m22738 s-22738		 26:m-10030 s-22738	27:m-10030 s-22738		 
28:m-10030 s10030	29:m22738 s10030		 30:m-10030 s10030	31:m-10030 s-22738		 

32:m22738 s10030	33:m-10030 s10030		 34:m-10030 s10030	35:m-10030 s10030		 
36:m22738 s-22738	37:m-10030 s-22738		 38:m22738 s10030	39:m-10030 s10030		 
40:m-10030 s-22738	41:m22738 s10030		 42:m22738 s-22738	43:m-10030 s-22738		 
44:m22738 s10030	45:m-10030 s10030		 46:m22738 s10030	47:m22738 s-22738		 

48:m22738
Extent=22738
Mark  min:-10030	 max:22738
Space min:10030	 max:10030

Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:42798
  Head: m22738  s42798
0:m22738 s-22738	1:m22738 s-22738		 2:m22738 s-22738	3:m22738 s-22738		 
4:m22738 s-22738	5:m22738 s-22738		 6:m22738 s-22738	7:m22738 s-22738		 
8:m22738 s-22738	9:m22738 s-22738		 10:m22738 s-22738	11:m22738 s-22738		 
12:m22738 s-22738	13:m22738 s-22738		 14:m22738 s-22738	15:m22738 s-22738		 

16:m22738 s-22738	17:m22738 s-22738		 18:m22738 s-22738	19:m22738 s-22738		 
20:m22738 s-22738	21:m22738 s-22738		 22:m22738 s-22738	23:m22738 s-22738		 
24:m22738 s-22738	25:m22738 s-22738		 26:m22738 s-22738	27:m22738 s-22738		 
28:m22738 s-22738	29:m22738 s-22738		 30:m22738 s-22738	31:m22738 s-22738		 

32:m22738 s-22738	33:m22738 s-22738		 34:m22738 s-22738	35:m22738 s-22738		 
36:m22738 s-22738	37:m22738 s-22738		 38:m22738 s-22738	39:m22738 s-22738		 
40:m22738 s-22738	41:m22738 s-22738		 42:m22738 s-22738	43:m22738 s-22738		 
44:m22738 s-22738	45:m22738 s-22738		 46:m22738 s-22738	47:m22738 s-22738		 

48:m22738
Extent=22738
Mark  min:22738	 max:22738
Space min:32767	 max:0

Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:42798
  Head: m22738  s42798
0:m22738 s-22738	1:m22738 s-22738		 2:m22738 s-22738	3:m22738 s-22738		 
4:m22738 s-22738	5:m22738 s-22738		 6:m22738 s-22738	7:m22738 s-22738		 
8:m22738 s-22738	9:m22738 s-22738		 10:m22738 s-22738	11:m22738 s-22738		 
12:m22738 s-22738	13:m22738 s-22738		 14:m22738 s-22738	15:m22738 s-22738		 

16:m22738 s-22738	17:m22738 s-22738		 18:m22738 s-22738	19:m22738 s-22738		 
20:m22738 s-22738	21:m22738 s-22738		 22:m22738 s-22738	23:m22738 s-22738		 
24:m22738 s-22738	25:m22738 s-22738		 26:m22738 s-22738	27:m22738 s-22738		 
28:m22738 s-22738	29:m22738 s-22738		 30:m22738 s-22738	31:m22738 s-22738		 

32:m22738 s-22738	33:m22738 s-22738		 34:m22738 s-22738	35:m22738 s-22738		 
36:m22738 s-22738	37:m22738 s-22738		 38:m22738 s-22738	39:m22738 s-22738		 
40:m22738 s-22738	41:m22738 s-22738		 42:m22738 s-22738	43:m22738 s-22738		 
44:m22738 s-22738	45:m22738 s-22738		 46:m22738 s-22738	47:m22738 s-22738		 

48:m22738
Extent=22738
Mark  min:22738	 max:22738
Space min:32767	 max:0

Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:42798
  Head: m22738  s42798
0:m22738 s-22738	1:m22738 s-22738		 2:m22738 s-22738	3:m22738 s-22738		 
4:m22738 s-22738	5:m22738 s-22738		 6:m22738 s-22738	7:m22738 s-22738		 
8:m22738 s-22738	9:m22738 s-22738		 10:m22738 s-22738	11:m22738 s-22738		 
12:m22738 s-22738	13:m22738 s-22738		 14:m22738 s-22738	15:m22738 s-22738		 

16:m22738 s-22738	17:m22738 s-22738		 18:m22738 s-22738	19:m22738 s-22738		 
20:m22738 s-22738	21:m22738 s-22738		 22:m22738 s-22738	23:m22738 s-22738		 
24:m22738 s-22738	25:m22738 s-22738		 26:m22738 s-22738	27:m22738 s-22738		 
28:m22738 s-22738	29:m22738 s-22738		 30:m22738 s-22738	31:m22738 s-22738		 

32:m22738 s-22738	33:m22738 s-22738		 34:m22738 s-22738	35:m22738 s-22738		 
36:m22738 s-22738	37:m22738 s-22738		 38:m22738 s-22738	39:m22738 s-22738		 
40:m22738 s-22738	41:m22738 s-22738		 42:m22738 s-22738	43:m22738 s-22738		 
44:m22738 s-22738	45:m22738 s-22738		 46:m22738 s-22738	47:m22738 s-22738		 

48:m22738
Extent=22738
Mark  min:22738	 max:22738
Space min:32767	 max:0

... (repeating last block indefinitely, only a power cycle / reset restores functionality at this point)

```

Note that when this occurs, turning the said lamp off (or toggling) doesn't change/fix this situation. So this is not a simple interference issue. Also I checked that is no electrical / EMI interference issue which directly affects the AVR or other board-level components and when issue happens rest of the code works normally, only IR decoding is broken. When this issue occurs, only a power cycle / reset restores functionality

Can you provide some pointers where to look at? Thanks.

michaldie, 2018-11-02:
I see a similar issue when I send a signal from my DENON remote RC1161
the first dump output looks reasonable (unknown) but it never stops thereafter
old IRremote does give me  :
Received unknown code, saving as raw
 m3300 s1800 m300 s500 m400 s450 m350 s1350 m300 s500 m350 s1350 m350 s500 m350 s1300 m350 s500 m350 s500 m350 s1300 m350 s500 m350 s500 m350 s1350 m350 s1300 m350 s500 m350 s500 m350 s450 m350 s500 m350 s500 m350 s500 m300 s500 m350 s1350 m350 s500 m350 s500 m300 s1350 m350 s500 m350 s500 m300 s500 m350 s1350 m350 s1350 m350 s450 m350 s1350 m350 s500 m350 s1300 m350 s500 m350 s500 m300 s500 m400 s450 m350 s500 m350 s500 m350 s1300 m400 s1300 m350 s500 m300 s500 m350 s1350 m350 s500 m300 s500 m350 s1350 m350

similar to Hazar, I am getting an endless stream of dump when triggering the effect.
I wonder if some buffer is overrun ?

Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:49428
  Head: m3350  s1700
0:m400 s450	1:m400 s450		 2:m350 s1300	3:m400 s450		 
4:m400 s1250	5:m400 s450		 6:m450 s1250	7:m400 s400		 
8:m400 s450	9:m450 s1250		 10:m400 s450	11:m400 s400		 
12:m400 s1300	13:m400 s1250		 14:m400 s450	15:m400 s450		 

16:m400 s450	17:m350 s500		 18:m350 s450	19:m400 s450		 
20:m400 s450	21:m400 s1250		 22:m400 s450	23:m400 s450		 
24:m400 s1250	25:m450 s400		 26:m400 s450	27:m400 s450		 
28:m400 s1250	29:m400 s1300		 30:m400 s450	31:m400 s1250		 

32:m400 s450	33:m400 s1300		 34:m350 s450	35:m450 s400		 
36:m400 s450	37:m400 s450		 38:m400 s400	39:m400 s450		 
40:m400 s1300	41:m400 s1250		 42:m400 s450	43:m400 s450		 
44:m400 s1250	45:m400 s450		 46:m400 s450	47:m400 s1250		 

48:m400
Extent=59100
Mark  min:350	 max:450
Space min:400	 max:1300

Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:46618
  Head: m36378  s19514
0:m19950 s22550	1:m19950 s22550		 2:m17450 s-486	3:m19950 s22550		 
4:m19950 s-2986	5:m19950 s22550		 6:m22450 s-2986	7:m19950 s20050		 
8:m19950 s22550	9:m22450 s-2986		 10:m19950 s22550	11:m19950 s20050		 
12:m19950 s-486	13:m19950 s-2986		 14:m19950 s22550	15:m19950 s22550		 

16:m19950 s22550	17:m17450 s25050		 18:m17450 s22550	19:m19950 s22550		 
20:m19950 s22550	21:m19950 s-2986		 22:m19950 s22550	23:m19950 s22550		 
24:m19950 s-2986	25:m22450 s20050		 26:m19950 s22550	27:m19950 s22550		 
28:m19950 s-2986	29:m19950 s-486		 30:m19950 s22550	31:m19950 s-2986		 

32:m19950 s22550	33:m19950 s-486		 34:m17450 s22550	35:m22450 s20050		 
36:m19950 s22550	37:m19950 s22550		 38:m19950 s20050	39:m19950 s22550		 
40:m19950 s-486	41:m19950 s-2986		 42:m19950 s22550	43:m19950 s22550		 
44:m19950 s-2986	45:m19950 s22550		 46:m19950 s22550	47:m19950 s-2986		 

48:m19950
Extent=1709766
Mark  min:17450	 max:22450
Space min:20050	 max:25050

....

michaldie, 2018-11-02:
going down the buffer trail - 
my lockup by DENON was fixed by increasing the buffer size .. 102 instead of 100 did the job...
IRLib2/IRLibGlobals.h:27:#define RECV_BUF_LENGTH 102

however it would seem anything that floods the buffer when using IRLibRecv.h would be able to cause this -

michaldie, 2018-11-02:
it seems it is IRLibRecv that struggles with any buffer overrun. IRLibRecvPCI does not show the same issue.