cyborg5/IRLib2 #23: Arduino 1.6.12 on WIN10 refused to install IRLib2-master.zib

MHz000, 2017-04-20:
I've donwloaded IRLib2-master.zib from gitHub and put it into my lib folder. Trying to add  it with ".zib lib include under sketches results in "The folder doesn't contain a valid .zip lib"

Arduino: 1.6.12 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

Angegebener Ordner/ZIP-Datei enthält keine gültige Bibliothek

Btw. is IRLib2 compatible with ESP8266?

Thank for any help

cyborg5, 2017-04-20:
I'm not sure what a ".zib" file is. When I download from GitHub I always get a ".zip" file. Then I unzip it, copy the five folders from the master folder into my Arduino library folder.

At this moment we do not support ESP 8266 but is probably the next feature we will be adding soon.

MHz000, 2017-04-20:
Chris
thank-you
.zib was a typo should be .zip  My experience with other .zip libraries was
just import them without unzipping. Arduino IDE will do the rest. Did this
successfull with IRremoteESP8266-master.zip.