cyborg5/IRLib2 #26: How to increase RAW_DATA length

karanmakharia, 2017-06-10:
I tried the Ken Shirriff's library but failed to control AC of different brands - LLOYD, DAIKIN, CARRIER etc. I tried your library - the raw recv and raw send examples. In the raw recv the length goes up to 100 and I am still not able to control my Air conditioner. I think that the length needs to be increased as there is more data coming from the remote. Can you tell me where is the length defined.

cyborg5, 2017-06-10:
The length is indeed probably your problem. Air conditioning protocols are notoriously very long. Look in IRLib2/IRLibGlobals.h line number 27

`#define RECV_BUF_LENGTH 100`

karanmakharia, 2017-06-11:
@cyborg5 It worked! But now the issue is each and every button has a different raw code of around 198 values. So is there a way to store those values? For all my Air conditioners the protocol is zero, so can I somehow make a protocol+hex combo to send data easily. 
There is one more issue in this: Each temperature value has a different hex code. So how to implement temperature low or high button - no matter what the temperature is.

cyborg5, 2017-06-11:
There is a section of the users manual that gives tips on how to create your own protocol but it requires pretty good knowledge of how IR protocols work and some pretty significant programming skills.

cameleon77, 2018-01-11:
@cyborg5 great library! I'm trying to analyze digital signal (not IR) and it works perfectly but only below 256 values, when I change RECV_BUF_LENGTH to e.g. 1000 it does not detect any signal any more (Arduino MEGA). Any other change is needed to go beyond 255? Thanks

cyborg5, 2018-01-18:
@cameleon77 I'm not able to re-create your issue. I've tried increasing the buffer length to 1000 on an Arduino Mega and had no problems. I also took the opportunity to double check the code to make sure that it is always indexed by a variable of type bufIndex_t which is fit to a values if the rings is 255 or less and 16 bit values if it is more. I don't have a source of signal that is more than 255 bytes long so I tried arbitrarily shortening the buffer length to just 50 so that it would overflow. The code worked okay. Of course it doesn't recognize the protocols because the end of the signal was truncated but it did detect a signal. I suppose there's a slim possibility that when the buffer overflows it may be overwriting a bite of data. My limit test could conceivably be off by one value. Perhaps there somewhere in the code I should have checked for greater than or equal to RECV_BUF_LENGTH-1. But I don't think so. Is your signal overflowing the buffer? Your description says that it somehow disables everything by increasing the buffer size and I just can't re-create that behavior.

projectkk2glider, 2018-01-20:
@cameleon77 I had the same issue when setting RECV_BUF_LENGTH bigger than 256 and I already found where the problem lies. I will post a PR soon to fix this.

closed 2018-01-20 by cyborg5

cameleon77, 2018-02-02:
Good catch, thanks for the prompt response :). Due to timing issues I had to move to ESP8266 and used some other library dedicated there. With ESP8266 got very stable readings so I now can log all OJ Electronics temperature sensors at home directly from the bus. I'll test the fix on my Arduino MEGA module once I have some time. Again, great job!

cameleon77, 2018-02-02:
OK, I did the test straight away and the fix works for me, kudos to @projectkk2glider !

PR #43 (2018-01-20): Fixes #26: Increasing RECV_BUF_LENGTH over 256 caused endless loop if…
… actual received number of samples was bigger than 256.

Closes #26
commit 0328cd7d (2018-01-20): Fixes #26: Increasing RECV_BUF_LENGTH over 256 caused endless loop if actual received number of samples was bigger than 256.
commit ef82c098 (2018-01-20): Merge pull request #43 from projectkk2glider/master

Fixes #26: Increasing RECV_BUF_LENGTH over 256 caused endless loop if…