cyborg5/IRLib2 #28: SAMD21 Compatibility with Serial.print

ckaraviotis, 2017-06-30:
IRLibDecodeBase uses Serial.print statements throughout to log information to the console.

In the SAMD21 M0 board, the method for printing to the USB console is exposed via SerialUSB.print.

As a result of this, when I tested any of the decode examples (using decoder.dumpResults) the arduino became unresponsive.

I found a quick fix by adding `#define Serial SERIAL_PORT_USBVIRTUAL` to the IRLibDecodeBase.cpp file, which makes all the print statements work OK.

This might need to be added to other files wrapped in an `#if`

cyborg5, 2017-06-30:
What do you mean specifically by "SAMD21 M0 board". We have tested the code on Arduino Zero, Arduino M0 Pro, Adafruit Feather M0 Basic, Adafruit Feather M0 BLE, Adafruit Feather M0 Express, Adafruit Circuit Playground Express all based on the SAMD21 processor and none of them needed the modification that you suggested. By the way these were all using the latest Arduino.cc IDE. What were you using to compile?

ckaraviotis, 2017-07-01:
This was using the Arduino M0 (not pro), with the latest version of the Arduino IDE, 1.8.2.

jdmarques, 2017-12-21:
I have the same problem using an Arduino M0 R3. According to what @ckaraviotis  said adding that line enables the examples which work as a receiver.
However, I can't get the sender to work properly. The sender example uploads but I can't receive anything. Is there any known change required for the sender?

jdmarques, 2017-12-21:
I had the same problem using an Arduino M0 R3. The fix that @ckaraviotis  suggested worked for the receiver example. 
However, I can't get the send example to function properly. The code runs but I don't receive anything. Is there any known change required for the send example to work in this Arduino?