cyborg5/IRLib2 #31: Arduino Air conditioner Remote Controller

krazbali, 2017-08-08:
With hashDecode.ino I got 2 codes, when I press one button. First is the header, its constant, but the second code is depend on the button (function). 
Serial monitor (power on button):
Unknown protocol. Hash value is: 0x811C9DC5
Unknownprotocol. Hash value is: 0x39FCE14C

The problem is I can't replay this codes...

cyborg5, 2017-08-08:
Most air conditioner codes are extremely long. You might need to increase the buffer size. Look in IRLibGlobals.h for the line

hectorlee, 2018-06-11:
Is there a buf length setting for sending with the rawSend sketch? The rawSend sketch isn't sending the full length of the ir code for a hitachi air conditioner remote but sends shorter codes just fine.

lrrrrr, 2018-06-12:
I am having the same issue as hectorlee. The length of my signal is 292, but rawSend sketch sends a much shorter signal.

lrrrrr, 2018-06-12:
Actually, there is an open pull request by alexalexein which resolves this. Successfully controlling a Mitsubishi air conditioner after implementing it.

cyborg5, 2019-09-20:
Because air-conditioners use extremely long sequences that are difficult to encode efficiently we will not be supporting air conditioner protocols.

closed 2019-09-20 by cyborg5