cyborg5/IRLib2 #32: Multiple definition when using TONE

scargill, 2017-08-12:
In a simple project which uses the TONE library (I'm using a NANO) I tried to compile in the IR receiver software and..

Tone.cpp.o (symbol from plugin): In function `timer0_pin_port':

(.text+0x0): multiple definition of `__vector_7'

libraries\IRLibRecv\IRLibRecv.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Nano.

cyborg5, 2017-08-12:
This is a conflict because most libraries use hardware timers. Try using the PCI receiver instead of the 50 µs interrupt receiver. Or determine which hardware timers are being used by the TONE library and change the default timers in IRLib2. You can find details in the documentation.

scargill, 2017-08-13:
Trying PCI

Note - Word manual - mistakes.... My - sometimes my.... and screen 7 - dumpResultso() should be dumpResults()

scargill, 2017-08-13:
Yup - PCI works. Ta.

closed 2017-08-13 by cyborg5

sb43201, 2017-08-31:
I'm still getting the same error using IRrecvPCI myReceiver(RECV_PIN); Could you show how exactly it should be done? Or how to change timer in IRLib2? I can't find the line that control the timers. Thanks!