cyborg5/IRLib2 #33: Problem by sending RAW signal

szmolnar, 2017-09-16:
I just started to play with the IRLib2.

I did an analyse on a Sony remote and decoded it successfully with analyze.ino
> Decoded Sony(2): Value:A50 Adrs:0 (12 bits)

Re-sending the code with send.ino works perfectly, TV is responding as expected
> mySender.send(SONY,0xa50, 12);

I did try a RAW sampling on the same command with rawRecv.ino, and I got
> #define RAW_DATA_LEN 26
uint16_t rawData[RAW_DATA_LEN]={
	2374, 622, 1170, 630, 574, 626, 1174, 622, 
	570, 626, 574, 626, 1178, 618, 574, 626, 
	1178, 618, 582, 618, 574, 626, 574, 622, 
	582, 1000};

When I try to re-send the captured data with rawSend the TV is not responding to the command.
The serial output tells that some data is send 2 times
> Sent signal.
Sent signal.

by re-capturing the raw data sent I'm getting a longer sampling LEN=2x26=52, something different like
> #define RAW_DATA_LEN 52
uint16_t rawData[RAW_DATA_LEN]={
	2394, 598, 1174, 654, 598, 622, 1118, 654, 
	598, 650, 562, 630, 1206, 622, 570, 650, 
	1170, 630, 574, 646, 594, 626, 554, 638, 
	586, 1106, 2418, 630, 1142, 686, 586, 634, 
	1198, 630, 582, 638, 586, 634, 1198, 658, 
	562, 630, 1174, 654, 570, 650, 558, 634, 
	538, 654, 590, 1000};

cyborg5, 2017-09-16:
Check out the documentation it explains that the standard for Sony protocol is that each frame of data should be sent three times. Also the Sony protocol defaults to 40 kHz rather than the standard 38 kHz used by most protocols. So if you are going to try to re-create a Sony signal using the raw data would need to modify the sketch to send 40 kHz and would need to send the signal three times consecutively in short order. The reason you're getting the signal sent 2 times probably has to do with whether or not you have return for return with linefeed set in your serial monitor. That may be causing double signals to be sent.

The raw format is really intended to be used only with unsupported protocols. As you can see it takes 26 words of data to represent the Sony signal using raw format. But because we know that it is Sony we can use a 16-bit word plus an eight bit value to store the number of bits. That's the reason we implement decoders and encoders is that we don't need to store all of that raw data. I presume you were just experimenting with the raw format to see how it works perhaps for use with unsupported protocols. But if it is supported protocol it's always best to use that protocol rather than the raw data.

szmolnar, 2017-09-24:
Thanks cyborg5!

Yes, I did play with raw to see how it is working.
I had no time to play with the Sony codes but I just quickly recorded an unsupported code and re-sending it is working.