cyborg5/IRLib2 #34: can't 'resend' a decoded signal

Boimb, 2017-09-22:
Hi,
I'm just starting using IRLib2 and wan't to emulate a remote (like most of users i presume).
I managed to decode some ir signals from my remote (ONKYO TX-SR508) with the IRrecv class. Could not manage to get results with IRRecvPCI.
Here's the serial output :
```

Decoded NEC(1): Value:4B36D32C Adrs:0 (32 bits) 
Raw samples(68): Gap:18474
  Head: m8850  s4450
0:m550 s600	1:m500 s1700		 2:m550 s550	3:m550 s550		 
4:m550 s1700	5:m550 s550		 6:m550 s1650	7:m550 s1700		 
8:m500 s600	9:m550 s550		 10:m550 s1650	11:m550 s1700		 
12:m550 s550	13:m550 s1650		 14:m550 s1700	15:m500 s600		 

16:m550 s1650	17:m550 s1650		 18:m550 s600	19:m550 s1650		 
20:m550 s550	21:m550 s550		 22:m550 s1700	23:m550 s1650		 
24:m550 s550	25:m550 s550		 26:m550 s1700	27:m550 s550		 
28:m550 s1650	29:m550 s1700		 30:m550 s550	31:m550 s550		 

32:m550
Extent=67100
Mark  min:500	 max:550
Space min:550	 max:1700
```
I tried then to emit this same code with the IRsendNEC but the device didn't respond.
I got a control LED on a receiver so i know some code is emitted. Seems to be the wrong format even if i filled the send function with the values obtained with the receiver.
```
IRsendNEC necSender;
necSender.send(0x4b36d32c, 32);
```
Any idea where i did wrong ?
Thx.
Boimb.

cyborg5, 2017-09-22:
Keep in mind that the PCI receive class only works on particular hardware pins.

Do you have 2 Arduinos where one is transmitting and the others receiving? What does the receiving Arduino see when you try to transmit? Some digital cameras such as WebCams can see an IR LED blinking. See if you can see anything coming out.

Boimb, 2017-09-22:
Thx for that quick reply.

> Keep in mind that the PCI receive class only works on particular hardware pins.

K. I'll take a closer look at the (really nice) doc. As i got results with the IRrecv class, i didn't pay more attention. Is it a big deal not to use IRrecvPCI ?

I have only 1 Arduino board, but as i said, i got an IrReceiver linked to a control led so I can check if IR is emitted... and it's blinking. So i presume code is emitted. Feels like the "protocol" or "encoding" is not the right one.
I've tried with raw too, and same thing... Control LED blinking, but device doesn't "understand" emitted code. 
I've read a similar issue [here](https://github.com/cyborg5/IRLib2/issues/33) about Sony device. The device only responds if code is sent multiple times and at a specific frequency.
Don't u think something similar could happen with ONKYOs ?
Cheers.