cyborg5/IRLib2 #35: Support for Adafruit Trinket - send only?

l8gravely, 2017-10-16:
I haven't been able to find a real answer yet, but would it be possible to use an Adafruit Trinket (ATTiny85) to implement a send-only version of this library?  I'm building a little box to turn on the sound bar when the TV gets turned on, and I'd like to use a Trinket as the controller.

https://learn.adafruit.com/introducing-trinket/introduction

Is what I'd like to use.  I know it's smaller in IO pins and memory, but it should be fast enough for what I need.

cyborg5, 2017-10-16:
I'm sorry we do not support the AT Tiny 85. I did recently purchase a Trinket M0 and we should be able to officially support. I just never gotten around to figuring out the details but because we support a variety of other M0 platforms it should be pretty trivial. I hadn't really thought about an output only support for the 85 but that might be possible. One of the problems with supporting input is the limited amount of memory for a buffer. However output might be possible. Unfortunately I just haven't worked on that.

l8gravely, 2017-10-16:
>>>>> "Chris" == Chris Young <notifications@github.com> writes:

Chris> I'm sorry we do not support the AT Tiny 85. I did recently
Chris> purchase a Trinket M0 and we should be able to officially
Chris> support. I just never gotten around to figuring out the details
Chris> but because we support a variety of other M0 platforms it
Chris> should be pretty trivial. I hadn't really thought about an
Chris> output only support for the 85 but that might be possible. One
Chris> of the problems with supporting input is the limited amount of
Chris> memory for a buffer. However output might be
Chris> possible. Unfortunately I just haven't worked on that.

Thanks for the quick reply!  I'll just have to keep looking around,
there are a couple of hints on the web for people who have made send
only setups work on the ATtiny85.

John

kgizzi, 2017-10-18:
I've subscribed to this thread. Can you please update this thread when support for the Trinket M0 is in place? I'm in need of it for a project! Thanks!