cyborg5/IRLib2 #37: Support for Trinket M0

lizcorson, 2017-10-28:
Would you be able to add support for the trinket M0? I was able to edit the library and get it working. In IRLibHardware.h, I duplicated the #elif defined (__SAMD21G18A__) cases and and used #elif defined (__SAMD21E18A__). I was able to use pin 1 on the trinket M0 and receive signals from an IR remote.

kgizzi, 2018-01-17:
@lizcorson I am unable to receive IR input by following your directions above for the Trinket M0. I've also tried duplicating the if statements for `__SAMD21G18A__` in IRLibSAM21.cpp and IRLibRecv.cpp as well but to no avail. If you could commit and reference your modified library with a description of the hardware (IR receiver and remote) used, it would be greatly appreciated!

lizcorson, 2018-01-24:
I used the Adafruit IR remote (https://www.adafruit.com/product/389) and receiver (https://www.adafruit.com/product/157). As for the code, I haven't used Github except for my own projects before. Is this the correct method to do that? http://kbroman.org/github_tutorial/pages/fork.html

cyborg5, 2018-01-24:
I am working on official support for the Trinket M0 and it will be available in a couple of days. I currently have input working on all available pins but I only get output to work on pins 0 and 2. Pin 1 does not support any kind of PWM output and will never be available. Pins 3 and 4 should work because they are similar to pins 8 and 9 on the Arduino Zero. But for some reason I cannot get them to work. I will be uploading the code whether I get pins 3 and 4 to work or not and perhaps someone else can figure out why they do not work. But before I upload these patches I want to do some testing on the Adafruit Gemma M0 as well. I will be working on that today.

closed 2018-01-24 by cyborg5