cyborg5/IRLib2 #41: RC5 detection repeats codes similar to a keybounce

jkanasu, 2017-12-30:
I am using the inbuilt "dump" example.
The code is modified is to show the issue.
I have a remove of "VU TV" remote model "ER-22645VU" which generates RC5 standard IR signals.
I press the keys only "once" but the program dumps it twice.
It seems like if we there is enough delay between the getResults and the enableIRIn method the issue does not show up.
See attached file image.
[image]

cyborg5, 2017-12-31:
If the dump routine is reporting two frames of data then that's what you're remote is sending. Putting in the delay just means that IRLib is missing that repeat frame. Notice that the toggle bit doesn't change during the repeats. That's how you detect separate keypresses. If the toggle bit has changed then you know it has been a press and release. If the toggle bit doesn't change then you know that the button was held down. Even though you are pressing the button very quickly, you're remote just decides it can go ahead and send multiple copies of the code. It's not an issue with the library. It's just the way your remote works.

jkanasu, 2018-01-01:
I agree with your assessment.
I assumed the remote would not generate code twice, may be it is generating. The next troubleshooting step is to use as oscilloscope to observe the waveforms generated by the IRreceiver (hardware).
Till then there is no reason to believe it is a IRLib issue.

closed 2018-01-01 by jkanasu