cyborg5/IRLib2 #47: Support for MKR1000

charifmahmoudi, 2018-02-21:
Could you please give me a hint to enable the support for MKR1000. i can try to add it and propose it a pull request.

Thanks

cyborg5, 2018-02-21:
We are already working on the Adafruit Feather M0 Wi-Fi which is extremely similar. Once we get that working we will be moving on to the MKR1000. There is already support for SAMD 21 based boards such as Feather M0, circuit playground express, Arduino Zero and so on. So it's just a matter of mapping pin numbers to the proper ports and putting in a couple of defines to recognize the board.

Look at IRLibProtocols/IRLibSAMD21.h if you really want to dig into the code yourself. That file is probably going to get a significant rewrite because we are supporting more and more M0 boards and it's getting a little complicated with all the conditional defines. We need a more systematic way to look up the port numbers.

cyborg5, 2018-03-10:
Support for the MKR1000 as well as the similar Adafruit Feather M0 Wi-Fi has now been added. See SAMD21.txt as well as examples/IoT_IR sample code.

closed 2018-03-10 by cyborg5