cyborg5/IRLib2 #49: Include problems between IRLib2 and IRLibProtocols

lhk, 2018-03-04:
I'm having a problem importing your library into my sketch.

I've downloaded a zip of the code and installed each folder individually as a library via the arduino IDE.
They are correctly (?) located in my sketchbook folder:
[image]

My code is minimalistic:
```
#include <IRLibAll.h>

void setup(){}
void loop(){}
```

Without the import, it compiles fine. With the import, it fails.
The error message is this:

> In file included from /home/lhk/sketchbook/libraries/IRLib2/IRLibAll.h:22:0,
>                  from sketch_mar04f.ino:1:
> /home/lhk/sketchbook/libraries/IRLib2/IRLibDecodeBase.h:13:28: fatal error: IRLibProtocols.h: No such file or directory
> compilation terminated.

Apparently, it can't import the IRLibProtocols.h file.
This is in fact not located in the IRLib2 folder, but in the IRLibProtocols.
How can I make the files see each other. I thought I had installed the libraries in the supposed way.

per1234, 2018-03-05:
I've explained the solution here:
https://stackoverflow.com/a/49101818

cyborg5, 2018-03-05:
Read the README file which says

This repository consists of a total of five libraries each of which must be in your arduino/libraries/ folder. So for example it should be installed as follows…

arduino/libraries/IRLib2
arduino/libraries/IRLibFreq
arduino/libraries/IRLibProtocols
arduino/libraries/IRLibRecv
arduino/libraries/IRLibRecvPCI
Do not install them in a single folder such as this…

arduino/libraries/IRLib2_master
IRLib2
IRLibFreq
IRLibProtocols
IRLibRecv
IRLibRecvPCI

closed 2018-03-05 by cyborg5