cyborg5/IRLib2 #51: Problem with sending codes on Pro Micro

JabberwockPL, 2018-03-18:
I have a Pro Micro with receiver on PIN 2 and sender on PIN 9. While I have no problem with receiving the codes, I cannot send anything - I need NECX and SAMSUNG36. The hardware is not a problem as it works with IRremote (but that one has no SAMSUNG36 support). Is anyone using Pro Micro? Maybe I am missing something? My code:
```

#include <IRLibSendBase.h>  
#include <IRLib_P07_NECx.h>
#include <IRLib_P08_Samsung36.h>  
#include <IRLibCombo.h>

IRsend mySender;

void setup() {
  Serial.begin(9600);
  delay(2000); while (!Serial); //delay for Leonardo
}

void loop() {
    mySender.send(SAMSUNG36,0x7887,0x0cb0);
    delay(500);
    mySender.send(NECX,0xe0e0e01f,0);
    Serial.println(F("Sent signal."));
    delay(1000);
}
```

JabberwockPL, 2018-03-18:
The code (and hardware) works fine when used with UNO on PIN 3. I am out of ideas...

cyborg5, 2018-03-18:
The Pro Micro I believe uses the 32u4 processor similar to the Arduino Leonardo. The deferral output pin on that processor is 9 instead of 3. Let me know if that works.

cyborg5, 2018-03-18:
My apologies. I see from your earlier message that you were using pin 9. I don't have a Pro Micro to test with. Can you make sure that the pin mapping really is the same as other 32u4 boards? Play around with the compiler defines in IRLibHardware.h and see if it is recognizing as a 32u4.

JabberwockPL, 2018-03-18:
I have tried to play around with the values (e.g. switching to timer2 with pin 5), but to no avail. I am not sure how can I check which defines are chosen in IRLibHardware.h?

JabberwockPL, 2018-03-19:
Hmm... I have moved to my 'production' (i.e. living-room) computer to test various options. Just to be sure I have compiled the code there and it works flawlessly... Maybe on my test computer the libraries were somehow borked.

So, for the information, Pro Micro works out of the box with the sender pin 9.

closed 2018-03-19 by JabberwockPL

cyborg5, 2018-03-19:
I'm glad you got it working