cyborg5/IRLib2 #53: Support for Adafruit WICED

todsms, 2018-04-17:
Hi,

Here's another question from the series "is it possible to support another platform?", but I decided to give it a try :)

I'm using Feather WICED for my AWS IoT project, and it took me under an hour to get it going with AWS. Unfortunately, now I need to enable IR transmitter functionality, but it looks like that IRLib2  does not support STM32 chip used in Feather WICED. Is there any chance to get it working?

Thanks,
Dennis

cyborg5, 2018-04-17:
That's probably a long way off if ever. Will do ESP 8266 and SAMD 51 (M4) before that. Note we did recently add an example using a Feather M0 Wi-Fi as well as a MKR1000. I don't know if either of those would work for you in place of the WICED. The advantage to the Feather M0 Wi-Fi is that the main processor is separate from the Wi-Fi processor. It makes it less likely that whatever I'm doing with IR is going to interfere with the connectivity.

todsms, 2018-04-17:
Hi Chris,

Thanks for the quick reply. I ended up using Feather WICED because Adafruit has built AWS IoT library compatible with Arduino, which I got working in under half hour. Feather WICED architecture also is using different chip for WiFi - Broadcom WICED, which supports TLS encryption (mandatory for AWS IOT).

closed 2018-04-18 by todsms