cyborg5/IRLib2 #55: Not working in setup function

b2cbd, 2018-06-01:
```
#include <IRLibSendBase.h>
#include <IRLib_P01_NEC.h> 
#include <IRLibCombo.h>

#define buttonCode 0xF7E01F

IRsend sendir;

void setup()
{
   // if i put it here, it doesn't work , no error 
   sendir.send(NEC,buttonCode,0);
}

void loop()
{
   // it works here!
   sendir.send(NEC,buttonCode,0);
}
```

what's wrong when I put it inside setup funciton?

cyborg5, 2018-06-03:
Were you able to resolve this on your own? I couldn't figure out what was wrong. Please tell us what you did in case someone else had a similar problem.

b2cbd, 2018-08-10:
It needed some delay after powering on the device, nothing related to this library. Thanks.

closed 2018-08-10 by b2cbd