cyborg5/IRLib2 #56: can't compile to MKR1000

WagnerMadeira, 2018-06-02:
**Using IDE 1.8.5 in Windows 10 and a bare minimum sketch:**

```
#include <IRLibAll.h>
void setup() {
}
void loop() {
  //Continue looping until you get a complete signal received
}

```**It compiles to other boards, like Aurduino/genuíno Uno, but to MKR1000, got erros bellow:**

In file included from C:\Users\wagne\OneDrive\Documents\Arduino\libraries\IRLibProtocols/IRLibHardware.h:83:0,
                 from C:\Users\wagne\OneDrive\Documents\Arduino\libraries\IRLib2\IRLibDecodeBase.cpp:11:
C:\Users\wagne\OneDrive\Documents\Arduino\libraries\IRLibProtocols/IRLibSAMD21.h:88:80: error: missing binary operator before token "18"
   #if ((IR_SEND_PWM_PIN > 21) || ( (IR_SEND_PWM_PIN > 12) && (IR_SEND_PWM_PIN  18)) )
                                                                                ^
C:\Users\wagne\OneDrive\Documents\Arduino\libraries\IRLibProtocols/IRLibSAMD21.h:89:12: warning: missing terminating " character [enabled by default]
     #error "Unsupported output pin on Arduino MKR 1000.
            ^
C:\Users\wagne\OneDrive\Documents\Arduino\libraries\IRLibProtocols/IRLibSAMD21.h:124:12: warning: missing terminating " character [enabled by default]
     #error "Unsupported output pin on Arduino M0 Pro
            ^
C:\Users\wagne\OneDrive\Documents\Arduino\libraries\IRLibProtocols/IRLibSAMD21.h:133:12: warning: missing terminating " character [enabled by default]
     #error "Unsupported output pin on Arduino Zero
            ^
C:\Users\wagne\OneDrive\Documents\Arduino\libraries\IRLibProtocols/IRLibSAMD21.h:189:0: warning: "IR_CLEAR_INTERRUPT" redefined [enabled by default]
 #define IR_CLEAR_INTERRUPT  IR_TCx->COUNT16.INTFLAG.bit.MC0 = 1;
 ^
In file included from C:\Users\wagne\OneDrive\Documents\Arduino\libraries\IRLib2\IRLibDecodeBase.cpp:11:0:
C:\Users\wagne\OneDrive\Documents\Arduino\libraries\IRLibProtocols/IRLibHardware.h:24:0: note: this is the location of the previous definition
 #define IR_CLEAR_INTERRUPT {} //clear interrupt flag
 ^
exit status 1
Erro compilando para a placa Arduino/Genuino 

**Can you help? Thanks a lot!**

WagnerMadeira, 2018-06-03:
Just included a "<" in #if ((IR_SEND_PWM_PIN > 21) || ( (IR_SEND_PWM_PIN > 12) && (IR_SEND_PWM_PIN 18)) ) 
and missing " at the end of #error  lines. Now it seems to be working.

cyborg5, 2018-06-03:
I'm not sure how this happened. My local copy was correct. Maybe something wrong during the upload. Anyway it's fixed now.

closed 2018-06-03 by cyborg5