cyborg5/IRLib2 #58: Can't compile Dump example on Arduino Uno

smd4587, 2018-09-05:
It appears the call to myDecoder.dumpResults is causing the compiler to error.  I get the following message.  If I comment out the call, it compiles fine.  Here is the output

lto1.exe: internal compiler error: in lto_output_varpool_node, at lto-cgraph.c:624

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

tgmaxx, 2018-10-11:
Same problem noted on Uno and Mega2560

1uuc4asb, 2018-10-31:
Same on Nano.

cyborg5, 2018-10-31:
This is a problem with the new updated AVR toolchain. Here is a link to a forum thread that explains a workaround. The body happens to any program that uses the F() macro to put character strings in program memory.
https://forum.arduino.cc/index.php?topic=574020.0

Nico04, 2018-12-01:
I've got the same issue with 1.6.23.
Reverting Arduino AVR to 1.6.21 work just fine 👍

ArminJo, 2019-02-08:
Hi,
changing the Pnames function to:
```
const __FlashStringHelper *Pnames(uint8_t type) {
    if (type > 89)
        return F("Unsup");
    if (type > LAST_PROTOCOL)
        type = UNKNOWN;
    // You can add additional strings before the entry for hash code.
    const __FlashStringHelper* const _Unknown = F("Unknown");
    const __FlashStringHelper* const _NEC = F("NEC");
    const __FlashStringHelper* const _Sony = F("Sony");
    const __FlashStringHelper* const _RC5 = F("RC5");
    const __FlashStringHelper* const _RC6 = F("RC6");
    const __FlashStringHelper* const _PanasonicOld = F("Panasonic Old");
    const __FlashStringHelper* const _JVC = F("JVC");
    const __FlashStringHelper* const _NECx = F("NECx");
    const __FlashStringHelper* const _Samsung36 = F("Samsung36");
    const __FlashStringHelper* const _G_I_Cable = F("G.I.Cable");
    const __FlashStringHelper* const _DirecTV = F("DirecTV");
    const __FlashStringHelper* const _rcmm = F("rcmm");
    const __FlashStringHelper* const _CYKM = F("CYKM");
//    const __FlashStringHelper* const _Additional_13 = F("Additional_13");
    const __FlashStringHelper *Names[LAST_PROTOCOL + 1] = { _Unknown, _NEC, _Sony, _RC5, _RC6, _PanasonicOld, _JVC, _NECx,
            _Samsung36, _G_I_Cable, _DirecTV, _rcmm, _CYKM
    //, _Additional_13//expand or edit these
            };
    return Names[type];
}
```
makes the compiler happy.

At least it is a workaround.