cyborg5/IRLib2 #60: Simple Sending/Receiving

Loremage, 2018-10-03:
I need to send a **small set of integers** over IR, say, {17, 1, 2, 3, 2, 1, 9}, **between two identical devices**.
(it could be a string if it has to be, or a set of characters, anything! I'll sort that out)

I've experimented with the rawSend and rawRecv functionality, as well as the Sony and NEC protocols, I've read much of the library documentation, forums, and the other issue requests.

It doesn't need to work with a television, or remote, or anything, just another of itself. I'm toggling the [.enableIRIn();], it can send, things, receive things, but never what was actually sent.

What is the the simple, reliable, way to encode a set of integers into timing logic... and decode them on the other side!?

I've been at this for several months now, what is the correct approach?

cyborg5, 2018-10-04:
Were you able to come up with a solution? I see you closed the issue. I would think using a simple protocol such as NEC you should be able to send anything up to 32 bits. Just don't use the ditto function of the NEC protocol.

Loremage, 2018-10-04:
I managed to get it to work poorly with the raw sending, and I closed it because I was worried this wasn't the right place for debugging.

That being said, you may have just led me to exactly what I needed:

This is the NEC power button code: (NEC,0x61a0f00f,0)
for fear of looking like an idiot, how the heck do I (properly) encode something else into that format!?

cyborg5, 2018-10-04:
Take a look at the "send.ino" sample code.

Loremage, 2018-10-04:
If you mean these boys, yeah, but they're both just hard coded to send a specific piece of data:

mySender.send(SONY,0xa8bca, 20);//Sony DVD power A8BCA, 20 bits
mySender.send(NEC,0x61a0f00f,0);//NEC TV power button=0x61a0f00f

The fundamental basic piece of knowledge that I was missing was that this "0x61a0f00f" is a hexadecimal value... I guess I've just never seen that before.

I can probably figure it out from here, thank you so much for your help!!

closed 2018-10-04 by Loremage