cyborg5/IRLib2 #61: "lto1.exe: internal compiler error" during compile

MajicDesigns, 2018-10-13:
**Environment**: 
Arduino Uno, IDE version 1.8.7, Arduino AVR boards Manager 1.6.23 (latest)

**IDE fault error message**:
lto1.exe: internal compiler error: in lto_output_varpool_node, at lto-cgraph.c:624
Please submit a full bug report, with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status
compilation terminated.
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Uno.

**Occurs**: 
Compile of my own code and also 'dump' and 'comboDump' example programs

**Description:**
The problem seems to be related to IRDecode. It started with my code when IRDecode was included. In comboDump.ino and combo.ino the problem disappears when the declaration and references to IRDecode are commented out.

MajicDesigns, 2018-10-13:
**Additional Info**:
Trying to just use 'send(), if we include all these header files individually
#include <IRLibRecvPCI.h>
#include <IRLibSendBase.h>
//#include <IRLib_P01_NEC.h>
//#include <IRLib_P02_Sony.h>
//#include <IRLib_P03_RC5.h>
//#include <IRLib_P04_RC6.h>
//#include <IRLib_P06_JVC.h>
//#include <IRLib_P07_NECx.h>
//#include <IRLib_P08_Samsung36.h>
//#include <IRLib_P11_RCMM.h>
#include <IRLibCombo.h>
Then as soon as one of the *_P??_* files is included the same error happens. Decode is nowhere referenced in this version of my code.

per1234, 2018-10-13:
See: https://github.com/arduino/ArduinoCore-avr/issues/39
A workaround is described at the above link.

The problem is specific to the avr-gcc 5.4.0 version so the alternative workaround is to use Boards Manager to roll back to Arduino AVR Boards 1.6.21.

MajicDesigns, 2018-10-13:
Rollback to AVR Boards 1.6.21 fixed the issue. Thanks.

closed 2018-10-13 by MajicDesigns