cyborg5/IRLib2 #63: IRLibSAMD51.h extern Tcc*  not being accepted by Arduino compiler

issuewithstuff, 2018-10-14:
I used this library in an Arduino project, but it did not compile, error message: 'Tcc' does not name a type
extern Tcc* = IR_TCCx;

If this is a stupid question, I am sorry. I am a total noob, but please tell me how to fix this.

issuewithstuff

cyborg5, 2018-10-15:
I think I know what the problem is and it's my fault. Give me a day or two. My off-line life is kind of complicated right now. For some reason IRLibSAM51.h is getting included even when you're not using a SAMD51 board.

issuewithstuff, 2018-10-15:
Ok. Thx.

cyborg5, 2018-10-17:
Can you give me some more specifics on this error? What type of board were you using when trying to compile? What versions of the Arduino IDE are you using?

issuewithstuff, 2018-10-21:
Arduino Mega 2560, ide 1.8.6

issuewithstuff, 2018-10-21:
code attached
[Universal_remote.zip](https://github.com/cyborg5/IRLib2/files/2499329/Universal_remote.zip)

issuewithstuff, 2018-11-06:
It's been awhile since I started this thread. Has this issue been resolved?

cyborg5, 2019-09-20:
All support for SAMD 51 and M4 boards has been recently updated. Let me know if you have further issues. SAMD51.txt has details. The main documentation will be updated soon.

closed 2019-09-20 by cyborg5