cyborg5/IRLib2 #74: Compile error on Grand Central M4

Alkerion, 2019-04-27:
The following error is raised when I try to compile on an Adafruit Grand Central M4

`Compiling 'ServerGC_M4' for 'Adafruit Grand Central M4 (SAMD51)'
 
IRLibSAMD51.cpp: In function void initializeSAMD51PWM(uint16_t)
 
IRLibSAMD51.cpp: 32:18: error: 'PIN_ATTR_PWM' was not declared in this scope
   if ( !((attr & PIN_ATTR_PWM) == PIN_ATTR_PWM) ) {
Error compiling libraries
Build failed for project 'ServerGC_M4'
`

cyborg5, 2019-04-27:
There was an update to the M4 core code and I haven't had a chance to update IRLib2 in order to accommodate it. Should have it working again in a week or so. Sorry for the inconvenience.

jspolsky, 2019-07-23:
quick hack to fix it:
#define PWM_ATTR_PWM (1UL<<3)

cyborg5, 2019-07-23:
I was aware that the M4 code was broken. I haven't had a chance to dig into it. I will check into your fix. Many thanks.

cyborg5, 2019-09-20:
All support for SAMD 51 and M4 boards has been recently updated. Let me know if you have further issues. SAMD51.txt has details. The main documentation will be updated soon.

closed 2019-09-20 by cyborg5