cyborg5/IRLib2 #78: Panasonic AC can't be controlled through IR

rajdpandey, 2019-05-23:
Hi all,

I was using Analysir' s Code for receiving and sending Long IR signal of AC but it isn't working. 

Tried this Issue: https://forum.arduino.cc/index.php?topic=418106.new#new

but didn't work.

Tried this also: https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

I have added the same issue here: https://irforum.analysir.com/viewtopic.php?f=6&t=963&p=1907#p1907

The result was the same no effect. So anyone Please Help me Regarding this.

cyborg5, 2019-09-20:
Because air conditioner remotes use extremely long sequences that are difficult to encode, we will not be supporting such protocols.

closed 2019-09-20 by cyborg5

rajdpandey, 2019-09-21:
@cyborg5 wow thanks for the response. Do you know how we can do that? means control AC from the Arduino using LongIR's? If you have any Idea please do tell.