cyborg5/IRLib2 #82: Cannot compile on Platformio; error: narrowing conversion

lou42, 2019-09-28:
line of errors when compiling for a trinket M0 on Platformio.

lib\IRLibProtocols\IRLibSAMD21.cpp:108:1: error: narrowing conversion of '(_ETCChannel)-1' from 'short int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
*** [.pio\build\adafruit_trinket_m0\lib7bb\IRLibProtocols\IRLibSAMD21.cpp.o] Error 1

[image]

cyborg5, 2019-09-29:
I cannot find this problem. I don't make use of "short int" in that file. And anyway "short int" should be an eight bit value anyway. The code compiles fine using the Arduino IDE. I'm not familiar with Platformio but it's something wrong with their definition of "short int".

lou42, 2019-09-29:
Thanks for looking into this.

[Platformio ](https://docs.platformio.org/en/latest/what-is-platformio.html) runs on Visual Studio Code and Atom and is a cross platform code builder and library manager for platforms like Arduino.

I searched through every file I am working with and 'short int' doesn't exist in any of my code either.

Regardless, I was able to get the code to compile by replacing all instances of 'uint8_t' in IRLibSAMD21.cpp with 'short int'.

Saberos, 2020-04-04:
Just ran into the same issue. My understanding of the Arduino ecosystem is still quite limited, but [here](https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/WVariant.h#L62) I get the impression the ETCChannel enum value NOT_ON_TIMER may be -1, which is assigned to the uint8_t field of the _IR_PWM_Map struct in IRLibSAMD21.cpp. Changing the IR_PER_EorF field to be of type short int resolved the error.
Appears to be related to using a SAMD Arduino board, personally using the MKR1000.

cyborg5, 2020-04-04:
@Saberos that is correct NOT_ON_TIMER is -1. For some reason the Arduino IDE doesn't flag that as an error. Could you try using "int8_t" instead of "uint8_t" or "short int" and tell me if it still works?

Saberos, 2020-04-14:
@cyborg5 changed IR_PER_EorF to `short int` as suggested by @lou42 and have successfully received and transmitted an IR signal with it.