cyborg5/IRLib2 #92: Buffers too short for Samsung Aircon

jpmeijers, 2020-02-04:
To get this library to learn and transmit my Samsung Aircon's OFF command I had to make a few modifications:

* In `IRLibGlobals.h` increase `RECV_BUF_LENGTH` to 400.
* In my main program, call `myReceiver.setFrameTimeout(100000);`. Shorter timeouts like 20000 was still too short.
* [BUG REPORT] In IRLib_HashRaw.h change the second parameter of the send function to be a uint16_t rather than a uint8_t, and change the for-loop to use `uint16_t i`.

The first two steps are mentioned in the documentation, but the third one is not. The third change can also not be changed easily using a configuration.

The raw data for my aircon's off command is:
```
#define RAW_DATA_LEN 350
uint16_t rawData[RAW_DATA_LEN]={
  570, 17848, 2985, 9044, 438, 561, 410, 1582, 
  411, 547, 447, 548, 452, 586, 411, 586, 
  409, 549, 448, 545, 451, 544, 479, 1513, 
  452, 545, 479, 547, 426, 1567, 446, 1546, 
  473, 565, 433, 1561, 411, 1582, 407, 1586, 
  410, 1583, 408, 1545, 449, 545, 455, 542, 
  477, 519, 479, 517, 481, 544, 426, 571, 
  426, 613, 409, 588, 408, 589, 408, 589, 
  409, 546, 451, 547, 449, 587, 409, 551, 
  446, 589, 410, 585, 409, 549, 449, 545, 
  475, 521, 452, 543, 454, 542, 481, 516, 
  482, 584, 384, 615, 383, 574, 445, 551, 
  447, 589, 407, 548, 477, 521, 451, 545, 
  449, 546, 452, 586, 409, 589, 409, 545, 
  451, 1542, 452, 1539, 480, 3062, 2916, 9001, 
  477, 1585, 382, 617, 381, 616, 405, 592, 
  408, 546, 477, 522, 473, 562, 411, 545, 
  450, 548, 449, 1558, 432, 550, 450, 542, 
  453, 1539, 453, 545, 475, 1518, 479, 1514, 
  477, 1547, 447, 1542, 476, 1518, 477, 1512, 
  480, 519, 452, 544, 453, 586, 409, 588, 
  407, 550, 449, 587, 409, 546, 449, 547, 
  452, 543, 452, 544, 478, 519, 480, 585, 
  386, 572, 424, 570, 477, 563, 408, 589, 
  434, 520, 452, 542, 456, 541, 454, 544, 
  451, 587, 412, 549, 446, 588, 407, 549, 
  449, 543, 453, 544, 455, 541, 481, 516, 
  478, 520, 477, 552, 421, 568, 426, 615, 
  407, 548, 473, 526, 476, 561, 408, 547, 
  451, 3089, 2914, 9001, 454, 1544, 449, 546, 
  450, 588, 408, 547, 449, 547, 450, 544, 
  481, 517, 479, 517, 481, 519, 449, 1566, 
  479, 562, 410, 547, 475, 1514, 480, 1557, 
  407, 1543, 453, 1540, 454, 586, 409, 1542, 
  452, 1539, 480, 1514, 479, 518, 479, 1540, 
  425, 577, 448, 1584, 434, 1514, 481, 560, 
  412, 587, 408, 548, 449, 1582, 409, 1547, 
  449, 1539, 451, 547, 451, 544, 479, 517, 
  480, 518, 474, 591, 383, 1611, 407, 1544, 
  479, 559, 437, 561, 409, 1584, 409, 1541, 
  451, 587, 411, 1540, 451, 1540, 481, 516, 
  481, 516, 481, 516, 481, 516, 477, 546, 
  425, 572, 452, 548, 446, 548, 476, 521, 
  477, 1557, 412, 1582, 410, 1000};
```

nestorix1343, 2020-06-18:
Found the same bug after some time debugging and wondering if I really could not control my Fujitsu airco system just before I found this bug report. 

Changing the type of the integer solved the problem for me also.

chandradharkoneti, 2023-01-18:
Thank you very very much, @jpmeijers 
This solved my issue.

rajSilikon, 2023-02-07:
Tested Raw Receive and Raw Send with my Samsung AC, works perfectly after following your instructions on the modifications.

just want to ask if its possible to use the hashDecode example with a longer data, and use that hash to send IR?

It just makes the data easier to handle; like storing in eeprom or sending through serial ( instead of hard coding the array).

I want to try to get it as 0x********* data, then I will create another arduino nano where in it will listen for serial input, and I can send that 0x********** for it to transmit IR.

rajSilikon, 2023-02-07:
[image]

Does this mean that I can not use hash code decoder if I want to send the IR again?

Im trying to create a universal remote that supports majority of the devices, and I found that the raw receive and raw send works with almost anything I tried. 

Other libraries can only support small data, and does not work on AC