cyborg5/IRLib2 #93: Can't get it to work with attiny85

maxholgasson, 2020-02-13:
Hi,
I can't get the library to work with an attiny 85 which has it's pwm on pin 5 (D0)

Where can I change the pin in the library and what's the advantage of this library compared to the original IRremote?

cyborg5, 2020-02-13:
I’m sorry but the library does not support AT tiny85.

skaman82, 2021-01-20:
Any chances the support will be added in the future? I am very interested in compiling the receive demo for the ATtiny85 but getting tons of errors.

ArminJo, 2021-01-21:
Try [IRMP](https://github.com/ukw100/IRMP). 
What are the problems with IRremote?

skaman82, 2021-01-30:
@ArminJo IR remote is working quite strange in my usecase. I am using an Apple remote and IRremote only detects a command when I place the remote within 10cm of my IR RX - otherwise nothing. IRLib2 works just fine.

ArminJo, 2021-02-03:
Does IRremote run better on a plain Arduino (UNO etc.)? It may depend on the setting of MARK_EXCESS_MICROS.