cyborg5/IRLib2 #99: IR Send Inconsistent at best, failing nearly 75% of the time, IR Recieve does not work at all

atomdog, 2020-10-13:
I've used the exact code from the Adafruit demo, running on an Arduino Pro Micro. Only difference is a 560 Ohm resistor rather than 470. I am able to successfully send commands around half the time, and only at specific angles. The original remote sends them from around 4x the distance. Seems like a software issue, code below, any idea what the problem is?
 if (beep == 1) {
    mySender.send(NEC,0xFFE21D, 32);
    delay(40);
    mySender.send(NEC,0xFFE21D, 32);
    delay(40);
    mySender.send(NEC,0xFFE21D, 32);
    
}
Using NEC protocol with a 32 bit message.

ArminJo, 2020-10-13:
Try [IRMP](https://github.com/ukw100/IRMP/tree/master/examples/OneProtocol) and/or another IR LED.

saabman, 2020-10-17:
I seem to be having erratic operational issues as well. Receiving and sending. 
I using an Arduino to receive the some codes from one remote and translate them to a set of codes for a different device. Some times it works great and other times its terrible Ive only just been looking at it the last couple of days and have not got any substantial leads yet. 

On the receiving side I have noticed I can get lots of "Unknown" decodes when no known transmitter is transmitting and have noticed it even with no IR receiver connected ?? Im leaning towards a hardware issue (chineseium UNO clone) So far Timing appears to be correct - at least for transmitting - investigation continuing...

atomdog, 2020-10-20:
I suspected timing to be an issue as well, but what is odd is that some receiving functionality that is lost with this library is regained by using other libraries. However, in those libraries sending does not work at all, as opposed to working around 75% of the time, so I do not think it's my Arduino. As much as it is poorly maintained repo's. I might just end up writing my own.