cyborg5/IRLib2 #120: IRLib2 wont compile when i try to compile for the Nano very

Icewhole, 2022-11-16:
The code compiles fine with an UNO but just wont compile when I use an Every, I need to be able to use the every as I am playing sounds after receiving an IR signal and the Every has the memory to do this.
Any thoughts ?.

Arduino: 1.8.18 (Windows 10), Board: "Arduino Nano Every, ATMEGA328

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\John_2\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\John_2\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\John_2\Documents\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=on -ide-version=10818 -build-path C:\Users\John_2\AppData\Local\Temp\arduino_build_440512 -warnings=none -build-cache C:\Users\John_2\AppData\Local\Temp\arduino_cache_866710 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\John_2\Documents\Arduino\libraries\IRLib2\examples\rawRecv\rawRecv.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\John_2\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\John_2\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\John_2\Documents\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=on -ide-version=10818 -build-path C:\Users\John_2\AppData\Local\Temp\arduino_build_440512 -warnings=none -build-cache C:\Users\John_2\AppData\Local\Temp\arduino_cache_866710 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\John_2\Documents\Arduino\libraries\IRLib2\examples\rawRecv\rawRecv.ino

Using board 'nona4809' from platform in folder: C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6

Using core 'arduino' from platform in folder: C:\Users\John_2\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6

Detecting libraries used...

"C:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10818 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\nona4809" "C:\\Users\\John_2\\AppData\\Local\\Temp\\arduino_build_440512\\sketch\\rawRecv.ino.cpp" -o nul

Alternatives for IRLibRecvPCI.h: [IRLibRecvPCI@2.0.19]

ResolveLibrary(IRLibRecvPCI.h)

  -> candidates: [IRLibRecvPCI@2.0.19]

"C:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10818 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\nona4809" "-IC:\\Users\\John_2\\Documents\\Arduino\\libraries\\IRLibRecvPCI" "C:\\Users\\John_2\\AppData\\Local\\Temp\\arduino_build_440512\\sketch\\rawRecv.ino.cpp" -o nul

Alternatives for IRLibRecvBase.h: [IRLib2@2.0.19]

ResolveLibrary(IRLibRecvBase.h)

  -> candidates: [IRLib2@2.0.19]

"C:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10818 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\nona4809" "-IC:\\Users\\John_2\\Documents\\Arduino\\libraries\\IRLibRecvPCI" "-IC:\\Users\\John_2\\Documents\\Arduino\\libraries\\IRLib2" "C:\\Users\\John_2\\AppData\\Local\\Temp\\arduino_build_440512\\sketch\\rawRecv.ino.cpp" -o nul

"C:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10818 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\nona4809" "-IC:\\Users\\John_2\\Documents\\Arduino\\libraries\\IRLibRecvPCI" "-IC:\\Users\\John_2\\Documents\\Arduino\\libraries\\IRLib2" "C:\\Users\\John_2\\Documents\\Arduino\\libraries\\IRLibRecvPCI\\IRLibRecvPCI.cpp" -o nul

Alternatives for IRLibHardware.h: [IRLibProtocols@2.0.19]

ResolveLibrary(IRLibHardware.h)

  -> candidates: [IRLibProtocols@2.0.19]

"C:\\Users\\John_2\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10818 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\John_2\\Ap [... cut ...]
C:\Users\John_2\Documents\Arduino\libraries\IRLibProtocols\IRLibSendBase.cpp:101:2: note: in expansion of macro 'IR_SEND_PWM_STOP'

  IR_SEND_PWM_STOP;

  ^~~~~~~~~~~~~~~~

Using library IRLibRecvPCI at version 2.0.19 in folder: C:\Users\John_2\Documents\Arduino\libraries\IRLibRecvPCI 

Using library IRLib2 at version 2.0.19 in folder: C:\Users\John_2\Documents\Arduino\libraries\IRLib2 

Using library IRLibProtocols at version 2.0.19 in folder: C:\Users\John_2\Documents\Arduino\libraries\IRLibProtocols 

exit status 1

Error compiling for board Arduino Nano Every.

SorcererSalamander, 2022-12-03:
I'm also trying to use IrLib2 with the Nano Every and getting pretty much the same error. I couldn't find this error message in my search until searching for this specific board and finding this post so I suspect it's a compatibility thing?

Icewhole, 2022-12-19:
I don't think the guy that wrote the library is making anymore updates as they don't seem to be replying for ages now.

cyborg5, 2022-12-19:
My apologies for not being more responsive. That particular board is not supported. The program will only compile our hardware that is in the supported list.

Icewhole, 2022-12-20:
Thank you Chris
Would you know of anyway of adding the new processor of the Every as the extra memory would really help in adding sounds etc. when a specific IR signal is detected.
Atm im stuck for physical space so only a Nano sized board would fit.

But thanks anyway for clearing that up as quite a few people are trying to use the Every with this library.

Kind regards

John