cyborg5/IRLib2 #122: AC Remote IR read issue

PaulAnurag, 2022-12-28:
Hi @cyborg5 
Thanks a lot for your work. I have been using your library for TV remotes and they are working good.

I also wanted to capture raw data using https://github.com/cyborg5/IRLib2/blob/master/IRLib2/examples/rawRecv/rawRecv.ino.
But every time I try to capture the data, my buffer shows the buffer length as 100 but I am pretty sure that the remote is sending more than that due to which when I am trying to send the captured data, it fails.

Can anyone help me on this issue?

chandradharkoneti, 2023-01-18:
I am having the same issue. It shows RAW_DATA_LEN 100. I have checked with a logic analyser. It shows length as 350.
Even in this video, https://youtu.be/9aLpvhuABK4?t=467 his output shows 350.

Please help. How do I increase the buffer length before the programme decides to call it quits??

chandradharkoneti, 2023-01-18:
https://github.com/cyborg5/IRLib2/issues/92

The above link is the solution to this issue.

PaulAnurag, 2023-01-18:
You have to follow the cpp files. You can extend the buffer.
Thanks.

closed 2023-01-18 by PaulAnurag