cyborg5/IRLib2 #124: Attiny 1604 support?

l8gravely, 2023-09-25:
Does anyone know if there will be support for the new AVR 8bit boards like the attiny16x4 or other newer members of the family?  I'm trying to migrate a project from a GemmaM0 board to a hand-made board using the attiny1604 but just ran into problems on my initial compile tests.  

I'm going to assume there's a bunch of work to be done to update timers and such.  Some of the errors are like this:

 In file included from /home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:14:0:
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp: In member function 'void IRsendBase::enableIROut(uint8_t)':
/home/john/Arduino/libraries/IRLibProtocols/IRLibHardware.h:292:34: error: 'TIMSK2' was not declared in this scope
  #define IR_RECV_DISABLE_INTR   (TIMSK2 = 0)
                                  ^
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:77:2: note: in expansion of macro 'IR_RECV_DISABLE_INTR'
  IR_RECV_DISABLE_INTR; //Timer2 Overflow Interrupt
  ^~~~~~~~~~~~~~~~~~~~
/home/john/Arduino/libraries/IRLibProtocols/IRLibHardware.h:292:34: note: suggested alternative: 'TIMERB2'
  #define IR_RECV_DISABLE_INTR   (TIMSK2 = 0)
                                  ^
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:77:2: note: in expansion of macro 'IR_RECV_DISABLE_INTR'
  IR_RECV_DISABLE_INTR; //Timer2 Overflow Interrupt
  ^~~~~~~~~~~~~~~~~~~~
/home/john/Arduino/libraries/IRLibProtocols/IRLibHardware.h:199:3: error: 'TCCR2A' was not declared in this scope
   TCCR2A = _BV(WGM20);  TCCR2B = _BV(WGM22) | _BV(CS20); \
   ^
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
  ^~~~~~~~~~~~~~~~~~
In file included from /home/john/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/avr/include/avr/io.h:99:0,
                 from /home/john/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/avr/include/avr/pgmspace.h:90,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/String.h:30,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/Print.h:24,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/Stream.h:25,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/Client.h:22,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/ArduinoAPI.h:29,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/Arduino.h:26,
                 from /home/john/Arduino/libraries/IRLibProtocols/IRLibProtocols.h:12,
                 from /home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.h:16,
                 from /home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:13:
/home/john/Arduino/libraries/IRLibProtocols/IRLibHardware.h:199:16: error: 'WGM20' was not declared in this scope
   TCCR2A = _BV(WGM20);  TCCR2B = _BV(WGM22) | _BV(CS20); \
                ^
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
  ^~~~~~~~~~~~~~~~~~
In file included from /home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:14:0:
/home/john/Arduino/libraries/IRLibProtocols/IRLibHardware.h:199:25: error: 'TCCR2B' was not declared in this scope
   TCCR2A = _BV(WGM20);  TCCR2B = _BV(WGM22) | _BV(CS20); \
                         ^
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
  ^~~~~~~~~~~~~~~~~~
In file included from /home/john/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/avr/include/avr/io.h:99:0,
                 from /home/john/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/avr/include/avr/pgmspace.h:90,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/String.h:30,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/Print.h:24,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/Stream.h:25,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/Client.h:22,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/ArduinoAPI.h:29,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/Arduino.h:26,
                 from /home/john/Arduino/libraries/IRLibProtocols/IRLibProtocols.h:12,
                 from /home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.h:16,
                 from /home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:13:
/home/john/Arduino/libraries/IRLibProtocols/IRLibHardware.h:199:38: error: 'WGM22' was not declared in this scope
   TCCR2A = _BV(WGM20);  TCCR2B = _BV(WGM22) | _BV(CS20); \
                                      ^
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
  ^~~~~~~~~~~~~~~~~~
/home/john/Arduino/libraries/IRLibProtocols/IRLibHardware.h:199:51: error: 'CS20' was not declared in this scope
   TCCR2A = _BV(WGM20);  TCCR2B = _BV(WGM22) | _BV(CS20); \
                                                   ^
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
[... 40 lines cut ...]
/home/john/Arduino/libraries/IRLibProtocols/IRLibHardware.h:196:31: error: 'TCCR2A' was not declared in this scope
  #define IR_SEND_PWM_STOP    (TCCR2A &= ~(_BV(COM2B1)))
                               ^
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:101:2: note: in expansion of macro 'IR_SEND_PWM_STOP'
  IR_SEND_PWM_STOP;
  ^~~~~~~~~~~~~~~~
In file included from /home/john/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/avr/include/avr/io.h:99:0,
                 from /home/john/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/avr/include/avr/pgmspace.h:90,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/String.h:30,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/Print.h:24,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/Stream.h:25,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/Client.h:22,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/api/ArduinoAPI.h:29,
                 from /home/john/.arduino15/packages/megaTinyCore/hardware/megaavr/2.6.8/cores/megatinycore/Arduino.h:26,
                 from /home/john/Arduino/libraries/IRLibProtocols/IRLibProtocols.h:12,
                 from /home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.h:16,
                 from /home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:13:
/home/john/Arduino/libraries/IRLibProtocols/IRLibHardware.h:196:47: error: 'COM2B1' was not declared in this scope
  #define IR_SEND_PWM_STOP    (TCCR2A &= ~(_BV(COM2B1)))
                                               ^
/home/john/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:101:2: note: in expansion of macro 'IR_SEND_PWM_STOP'
  IR_SEND_PWM_STOP;
  ^~~~~~~~~~~~~~~~
exit status 1
Error compiling for board ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204.

cyborg5, 2023-09-25:
We have no plans to support additional 8-bit boards. There are so many advanced boards 16 and 32 bit varieties that are coming out constantly and we are struggling to find the time to add support for them. I can't see us going back to 8-bit.

You could try to implement a solution yourself. There are three hardware specific things needed by IRLib2. If you want to use the 50 µs interrupt timer you need to figure out how to do that on your board. That polls the input pin every 50 µs to see if it is on or off. An easier way to do it is if your board supports a pin change interrupt. If so, it is easier to use than the 50 µs timer. It occasionally fails if you are sending multiple signals rapidly but overall it's a better solution.

That covers receiving. The other hardware specific thing you need to do is you need to be able to generate a PWM frequency ranging from about 30 kHz up to 56 kHz with a duty cycle approximately 60% on and 40% off. That is extremely hardware specific and most Arduino implementations of different chips do not provide built-in functions to change PWM frequencies. If they did, it would be extremely easy to port this code to different platforms.

I'm sorry I don't have better news for you. Good luck in all you do.

closed 2023-09-25 by cyborg5

l8gravely, 2023-09-29:
Thanks for the reply and the info.  Since the attiny16x4 model is 8bit, and quite alot like other attiny chips, I'll try to see if I can make it work.  A good learning experience for sure.   Looking at your code, I should just have to add in the right ifdefs to setup things properly.  Famous last words!